cantools/cantools #687: cantools.database.decode_message() failed to decode message data with length > 8 characters (e.g. CAN-FD message)

luojun6, 2024-08-21:
Hello,

I am trying to use python-can and cantools to decode real-time CAN bus asc or blf trace. While I am decoding a CAN-FD message, it was found that cantools.database.decode_message() could not decode message with length > 8 characters.

```python
dbc = cantools.database.load_file("my_CANFD.dbc")
trace = list(can.BLFReader('my_trace.blf'))

for msg in trace:
    if msg.arbitration_id == 550:
        print(msg)
        print(dbc.decode_message(msg.arbitration_id, msg.data, decode_containers=True, scaling=False))
```

The output is:
```
Timestamp: 1723091311.296079    ID:      226    S Rx     F BS       DL: 48    00 02 25 03 d8 65 c8 00 02 89 08 26 60 00 00 00 00 00 00 00 02 02 08 00 00 00 00 0a 50 f6 68 00 02 8a 08 00 00 00 00 00 00 15 d2 00 00 00 00 00    Channel: 0
{'Header_ID': 549, 'Header_DLC': 3, 'signal_0': 216, 'signa_l': 6, 'signal_2': 'value_xxx', 'signal_3': 'value_yyy', 'signal_4': 'value_zzz', 'signal_5': 'value_abc', 'signal_6': 'value_def'}
Timestamp: 1723091311.296079    ID:      226    S Rx     F BS       DL: 48    00 02 25 03 d8 65 c8 00 02 89 08 26 60 00 00 00 00 00 00 00 02 02 08 00 00 00 00 0a 50 f6 68 00 02 8a 08 00 00 00 00 00 00 15 d2 00 00 00 00 00    Channel: 1
{'Header_ID': 549, 'Header_DLC': 3, 'signal_0': 216, 'signa_1': 6, 'signal_2': 'value_xxx', 'signal_3': 'value_yyy', 'signal_4': 'value_zzz', 'signal_5': 'value_abc', 'signal_6': 'value_def'}
```

You can see the raw data of message contains probably 32 characters (a normal CAN-FD message), but cantools only could decode <= 8 characters. But if I used `dbc.get_message_by_frame_id(550).signals`, I can see all of the decoded signal definitions.

Very sorry that I can't share the testing blf trace and dbc, because it's confidential materials.

luojun6, 2024-08-21:
> Hello,
>
> I am trying to use python-can and cantools to decode real-time CAN bus asc or blf trace. While I am decoding a CAN-FD message, it was found that cantools.database.decode_message() could not [...]

In addition, actually the dbc file is converted from a arxml file by `canmatrix`, some PDU information (e.g. container) had been missed. When I check the message, the `is_container` attribute is `False`, maybe this is the reason of decoding error.

[image]

But unfortunately, I can't read the arxml edited by Vector CANoe. It raised a ValueError as `Floating point value specified where integer is required` while I was trying to add_arxml_file.

andlaus, 2024-08-22:
>  In addition, actually the dbc file is converted from a arxml file by canmatrix, some PDU information (e.g. container) had been missed. When I check the message, the is_container attribute is False, maybe this is the reason of decoding error.

as far as I now DBC does not support AUTOSAR containers (or at least, cantools does not implement it), so `.is_container` is always false for databases stemming from DBC files...

andlaus, 2024-08-22:
> But unfortunately, I can't read the arxml edited by Vector CANoe. It raised a ValueError as Floating point value specified where integer is required while I was trying to add_arxml_file.

can you provide a backtrace?