cantools/cantools #691: Building wheel for bitstruct (pyproject.toml) did not run successfully

Paolo-Beci, 2024-09-09:
When I install cantools dependencies using `python3 -m pip install -e '.[dev]'` command I get the following error:

> Building wheels for collected packages: cantools, bitstruct
>   Building editable for cantools (pyproject.toml) ... done
>   Created wheel for cantools: filename=cantools-0.1.dev1738+gc8a66a9-0.editable-py3-none-any.whl size=7220 sha256=53ea8a66d0695be982ebda11206cc60e9d922964cc230bd67de23575700e856b
>   Stored in directory: /private/var/folders/73/fnbj0zlj00766bj81z10mvgh0000gn/T/pip-ephem-wheel-cache-rh8jalgw/wheels/32/08/b2/f77f0277534636b539fb9cc33578fb9690b6110e46e37e1995
>   Building wheel for bitstruct (pyproject.toml) ... error
>   error: subprocess-exited-with-error
>   
>   × Building wheel for bitstruct (pyproject.toml) did not run successfully.
>   │ exit code: 1
>   ╰─> [29 lines of output]
>       running bdist_wheel
>       running build
>       running build_py
>       creating build
>       creating build/lib.macosx-14.0-arm64-cpython-312
>       creating build/lib.macosx-14.0-arm64-cpython-312/bitstruct
>       copying src/bitstruct/__init__.py -> build/lib.macosx-14.0-arm64-cpython-312/bitstruct
>       running egg_info
>       writing src/bitstruct.egg-info/PKG-INFO
>       writing dependency_links to src/bitstruct.egg-info/dependency_links.txt
>       writing top-level names to src/bitstruct.egg-info/top_level.txt
>       reading manifest file 'src/bitstruct.egg-info/SOURCES.txt'
>       reading manifest template 'MANIFEST.in'
>       adding license file 'LICENSE'
>       writing manifest file 'src/bitstruct.egg-info/SOURCES.txt'
>       copying src/bitstruct/bitstream.c -> build/lib.macosx-14.0-arm64-cpython-312/bitstruct
>       copying src/bitstruct/bitstream.h -> build/lib.macosx-14.0-arm64-cpython-312/bitstruct
>       copying src/bitstruct/c.c -> build/lib.macosx-14.0-arm64-cpython-312/bitstruct
>       running build_ext
>       building 'bitstruct.c' extension
>       creating build/temp.macosx-14.0-arm64-cpython-312
>       creating build/temp.macosx-14.0-arm64-cpython-312/src
>       creating build/temp.macosx-14.0-arm64-cpython-312/src/bitstruct
>       clang -fno-strict-overflow -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -O3 -Wall -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk -I /opt/homebrew/opt/qt@5/include -I/Users/paolob/Developer/squadracorse/cantools/venv/include -I/opt/homebrew/opt/python@3.12/Frameworks/Python.framework/Versions/3.12/include/python3.12 -c src/bitstruct/bitstream.c -o build/temp.macosx-14.0-arm64-cpython-312/src/bitstruct/bitstream.o
>       clang -fno-strict-overflow -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -O3 -Wall -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk -I /opt/homebrew/opt/qt@5/include -I/Users/paolob/Developer/squadracorse/cantools/venv/include -I/opt/homebrew/opt/python@3.12/Frameworks/Python.framework/Versions/3.12/include/python3.12 -c src/bitstruct/c.c -o build/temp.macosx-14.0-arm64-cpython-312/src/bitstruct/c.o
>       clang -bundle -undefined dynamic_lookup -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk /opt/homebrew/opt/qt@5/lib -I /opt/homebrew/opt/qt@5/include build/temp.macosx-14.0-arm64-cpython-312/src/bitstruct/bitstream.o build/temp.macosx-14.0-arm64-cpython-312/src/bitstruct/c.o -o build/lib.macosx-14.0-arm64-cpython-312/bitstruct/c.cpython-312-darwin.so
>       ld: file cannot be mmap()ed, errno=22 path=/opt/homebrew/opt/qt@5/lib in '/opt/homebrew/opt/qt@5/lib'
>       clang: error: linker command failed with exit code 1 (use -v to see invocation)
>       error: command '/usr/bin/clang' failed with exit code 1
>       [end of output]
>   
>   note: This error originates from a subprocess, and is likely not a problem with pip.
>   ERROR: Failed building wheel for bitstruct
> Successfully built cantools
> Failed to build bitstruct
> ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (bitstruct)
> 

Environment: macosx-14.0-arm64

andlaus, 2024-09-09:
if I interpret this correctly, your C compiler (clang) complains that it cannot link to Qt:
```
clang -bundle -undefined dynamic_lookup -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk /opt/homebrew/opt/qt@5/lib -I /opt/homebrew/opt/qt@5/include build/temp.macosx-14.0-arm64-cpython-312/src/bitstruct/bitstream.o build/temp.macosx-14.0-arm64-cpython-312/src/bitstruct/c.o -o build/lib.macosx-14.0-arm64-cpython-312/bitstruct/c.cpython-312-darwin.so
ld: file cannot be mmap()ed, errno=22 path=/opt/homebrew/opt/qt@5/lib in '/opt/homebrew/opt/qt@5/lib'
```

since `bitstruct.c` most definitely does not use Qt, this seems to be an issue with your setup...

Paolo-Beci, 2024-09-10:
Solved, thanks!

closed 2024-09-10 by Paolo-Beci