cantools/cantools #692: How to know what message a signal belongs to?

C47D, 2024-09-11:
Hi,

I been using cantools to inspect some DBC files, in my particular use case I need to display some signal properties, including the frame it belongs to, but I haven't been able to figure out a way to get that information. Is there an API that can help me figure out that?

andlaus, 2024-09-11:
there's currently no build-in way, but given that in order to access a signal object, one needs to decode a message, it did not occur to me that this could be required. if you *really* need this functionality, I'd advise to just go over all messages in the database and find the one which exihibits the name of the signal in question (note that there might be multiple messages containing the same signal);
```python
def find_messages_with_signal(db, signal_name):
  result = []
  for msg in db.messages:
    for sig in msg.signals:
      if sig.name == signal_name:
        result.append(msg)
  return result
```

(if you have a database featuring container messages, you also have to check the signals of `.contained_messages`.)

C47D, 2024-09-16:
Thanks for the snippet, I'll try it out!

closed 2024-09-16 by C47D