cantools/cantools #709: Only first note of KCD object gets captured

Phylex, 2024-10-25:
when Importing a KCD, the contents of the `<Notes>` field is placed into the 'comments' dict of the message class. If multiple notes are present, only the first note is passed into the comments dictionary of the message. It would be nice if all notes would be placed into the comments dictionary with an integer index as a key, as the KCD file spec does not allow for attributes on the notes field.

andlaus, 2024-10-27:
agreed. can you open a PR? be aware that currently the key for `.comments` is intended to be the language, so maybe it is better to simply concatenate the `<Notes>`?

Phylex, 2024-10-28:
So I would prefer if the output where a list of notes as the value to the dict then. 
I was planning on using the Notes field to encode non-standard properties of the parent object of the note, thus I'd like separate notes to be separate strings.
Does KCD allow for the specification of the language in which it is written? I don't see that in the KCD parsing code.