cantools/cantools #718: Change source and header format for C source generation to be usable by doxygen without EXTRACT_ALL

felix-khlmnn, 2025-01-04:
Hello everybody,

I have started to integrate cantools into my workflow for developing a project I'm currently working on.
For this project, I only need the C source generator capabilities of this project. The generated source and header files are a great help to keep everything tidy and to reflect any changes made to the database at a later date in my code.

The project I am working on also uses Doxygen to generate its documentation. I have added the generated `.c` and `.h` files to the Doxygen input and realised that they weren't documented unless the `EXTRACT_ALL` setting was activated in the Doxyfile. This is due to the file metadata in the beginning of the file not being present/correctly formatted.

I think it would make more sense to make the file metadata Doxygen-"compliant" in order to integrate its use into a workflow more easily.

If this is a sensible addition from your perspective, I'll happily adopt the template and create a corresponding pull request.

Best regards
Felix Kuhlmann

andlaus, 2025-01-13:
I see no harm in changing the output of the code generator in such more or less stylistic ways to accommodate doxygen (I hope I understood your proposal correctly). IOW, as long as your PR does not add a command line option to the code generator, I'll very likely happy to merge it.

felix-khlmnn, 2025-01-15:
Alright, thank you. I will look into it in the next few days and reference this issue in the PR.

PR #720 (2025-01-15): Make generated src files more Doxygen-compatible
The templates for the source and header files lacked the file meta information needed by Doxygen to display the files in its file listing.

Until now, `EXTRACT_ALL` was needed to extract the Doxygen tags provided by the generated files. This is now fixed.

Resolves: #718
commit 584a67fb (2025-01-15): Make generated src files more Doxygen-compatible

The templates for the source and header files lacked the file meta
information needed by Doxygen to display the files in its file listing.

Until now, `EXTRACT_ALL` was needed to extract the Doxygen tags provided
by the generated files. This is now fixed.

Resolves: #718
closed 2025-01-15 by andlaus