cantools/cantools #725: EncodeError when init value is bigger than maximum value

sgnes, 2025-02-15:
### The issue:
When init value is bigger than maximum value, it will raise an EncodeError in tester

[image]

[image]

### Analysis:
in _prepare_initial_signal_values if initial exist, it will directly use it without any check.
`
    def _prepare_initial_signal_values(self):
        initial_sig_values = {}

        # Choose a valid set of mux settings
        mplex_settings = {}
        for m0 in reversed(self._mplex_map.values()):
            for m1 in m0:
                mplex_settings.update(m1)

        for signal in self.database.signals:
            minimum = 0 if not signal.minimum else signal.minimum
            maximum = 0 if not signal.maximum else signal.maximum
            if signal.initial:
                # use initial signal value (if set)
                initial_sig_values[signal.name] = signal.initial
            elif signal.is_multiplexer:
                initial_sig_values[signal.name] = mplex_settings.get(signal.name, 0)
            elif minimum <= 0 <= maximum:
                # use 0 if in allowed range
                initial_sig_values[signal.name] = 0
            else:
                # set at least some default value
                initial_sig_values[signal.name] = minimum
        return initial_sig_values
`
in _update_can_message it did not pass "strict" to database.encode, so encode will use default which is True
then it will raise an error in this encode function.

```
def _update_can_message(self):
        arbitration_id = self.database.frame_id
        extended_id = self.database.is_extended_frame
        pruned_data = self.database.gather_signals(self.data)
        data = self.database.encode(pruned_data,
                                    self.scaling,
                                    self.padding)
        self._can_message = can.Message(arbitration_id=arbitration_id,
                                        is_extended_id=extended_id,
                                        data=data)

    def encode(self,
               data: EncodeInputType,
               scaling: bool = True,
               padding: bool = False,
               strict: bool = True,
               ) -> bytes:

```

### Proposal:

After check the code, at least we have two proposals, pls let me know your comments, tks

#### Add check in _prepare_initial_signal_values

[image]

#### Add an optional parameter "strict" 

this will require to add this parameter to below class/function, so that user can have control to encode function

1. cantools.tester.Tester
2. Message
3. Message.update
4. Message._update_can_message

### Why need this change
Sometimes in the DBC file there will be  a lot of such kind of issue, fix one by one will take time, so it would be better if user can have control whether raise error for such issue.

zariiii9003, 2025-02-15:
Is the initial value mentioned in the value table?

sgnes, 2025-02-15:
> Is the initial value mentioned in the value table?

init value isdefined by GenSigStartValue, you can check the dbc file I had uploaded, tks.

zariiii9003, 2025-02-15:
Usually the initial value either corresponds to a valid physical value (after scaling), or a value in the value table. Otherwise it has no meaning...

sgnes, 2025-02-16:
> Usually the initial value either corresponds to a valid physical value (after scaling), or a value in the value table. Otherwise it has no meaning...

you are right, I would not say this is a issue of cantools, this out of ranges for sure a issue of the dbc file.
sometimes the DBC file is comefrom customer, not easy/allowed to modify it.
So would be great if user can have control of whether raise exceptions for such issue, thanks.

commit c02434f3 (2025-02-16): add a strict par to fix #725
commit fa06c208 (2025-02-16): Merge pull request #726 from sgnes/EncodeError#725

add a strict par to fix #725
closed 2025-02-16 by andlaus