cantools/cantools #728: error when import cantools

pbarcaFerrari, 2025-03-12:
Hi at alls,

when in python console import cantools i have this error:

 70     """Return a mapping of signals to the multiplex settings that will
     71     yield the signal.
     72 
     73     {signal: [{mplexers}, {mplexers...}]}
     74 
     75     """
     76     return _invert_signal_tree(tree)
---> 78 class Listener(can.Listener):
     80     def __init__(self, database, messages, input_queue, on_message):
     81         self._database = database

AttributeError: module 'can' has no attribute 'Listener'

I use python ver 3.13.
Many thanks

zariiii9003, 2025-03-12:
You probably have a module named `can.py` that is shadowing the `can` library (python-can).

pbarcaFerrari, 2025-03-12:
Many thanks,
I resolve unistalling can library and python-can library and then reinstall python-can library.

closed 2025-07-06 by zariiii9003