cantools/cantools #735: problem in autogenerating c source code from dbc file if endiannes changed

VP1701, 2025-05-05:
I have this dbc file for a sensor:

```
VERSION "encoder DBC V 1.0"

BS_:

BU_: ARC_SENSOR

BO_ 2566851456 ARC_SENSOR_ANGLE: 8 ARC_SENSOR
    SG_ Angle1 : 0|16@1+ (0.1,0) [0|360] "deg" ARC_SENSOR 
    SG_ Angle2 : 16|16@1+ (0.1,0) [0|360] "deg" ARC_SENSOR
    SG_ Reserved : 32|16@1+ (1,0) [255|255] "" ARC_SENSOR
    SG_ Error : 48|8@1+ (1,0) [0|255] "" ARC_SENSOR
    SG_ Checksum : 56|8@1+ (1,0) [0|255] "" ARC_SENSOR

```

I wanted to change the endianness it has. so I modified it to be

```
VERSION "encoder DBC V 1.0"

BS_:

BU_: ARC_SENSOR

BO_ 2566851456 ARC_SENSOR_ANGLE: 8 ARC_SENSOR
    SG_ Angle1 : 0|16@0+ (0.1,0) [0|360] "deg" ARC_SENSOR 
    SG_ Angle2 : 16|16@0+ (0.1,0) [0|360] "deg" ARC_SENSOR
    SG_ Reserved : 32|16@0+ (1,0) [255|255] "" ARC_SENSOR
    SG_ Error : 48|8@0+ (1,0) [0|255] "" ARC_SENSOR
    SG_ Checksum : 56|8@0+ (1,0) [0|255] "" ARC_SENSOR

```

but now the auto generation of c source code fails and gives error:
```
python3 -m cantools generate_c_source encoder_0.dbc
error: DBC: "The signal Checksum does not fit in message ARC_SENSOR_ANGLE."
```
What did I do incorrectly? Does the endianness affect the amount of bits or where the bit reading is started?

juleq, 2025-05-05:
You have changed the endianness without changing the start bit. The start bit will be different for a signal that you change the endianness for. Little tricky part of the dbc spec. I would recommend you install CANdb++ (should be included in Vector demo packages) and change it there. It will translate the start bit appropriately.

juleq, 2025-05-05:
Startbits for big endian would be 7, 23, 39, 55 and 63.

VP1701, 2025-05-06:
Thank you I flipped the start bits and now it works.

closed 2025-05-06 by VP1701