cantools/cantools #736: **Support decoding J1939/ISOBUS frames from candump: decouple PGN logic from strict CAN ID mapping**

oak-tree, 2025-05-13:
## 🧠 Background

The `cantools` library currently assumes a strict mapping between CAN IDs and messages as defined in the DBC. This works well for standard CAN, but not for **J1939** or **ISOBUS**, where:

* The **CAN ID includes a combination** of **Priority**, **PGN**, **Source Address**, and sometimes **Destination Address**.
* The **PGN** itself is extracted from the 29-bit CAN ID and may or may not include a destination address depending on its **format**.

The decoding logic in `cantools` expects the full CAN ID to match exactly what's in the DBC, which makes decoding `candump` logs for **dynamic J1939 traffic** difficult.

---

## 🧪 Example Problem

A J1939 frame might look like this (from `candump`):

```
(1601234567.123456) can0 18FEF100#1122334455667788
```

Breakdown of CAN ID `0x18FEF100`:

* Priority: `0x6`
* PGN: `0xFEF1`
* Source Address: `0x00`

### Problem:

`cantools` expects `0x18FEF100` as a hardcoded CAN ID in the DBC, and won’t decode it if only `PGN 0xFEF1` exists in the DBC, even though `0x18FEF1XX` would be valid for multiple sources.

This becomes more complex with:

* **PDU1 format PGNs (0x00–0xEF)** → use Destination Address
* **PDU2 format PGNs (≥0xF0)** → Destination field becomes part of PGN

Hence, the **CAN ID** will vary depending on the source and destination, but **PGNs remain the same** — and the DBC should allow for general matching on PGNs, not exact CAN IDs.

---

## 📌 Feature Request / Proposed Solution

* Add an option to decode based on **PGN + message matching**, **ignoring source/destination** unless explicitly needed.
* Support decoding **J1939/ISOBUS logs** (e.g., from `candump`) by **extracting the PGN** from the extended CAN ID, then matching it to the message definitions in the DBC.
* Optionally, expose helper functions to:

  * Parse 29-bit CAN IDs into PGN, SA, DA.
  * Match PGNs from the DBC regardless of SA.

---

## 💬 Example API Suggestion

```python
cantools.decode_j1939_frame(frame_id, data, dbc, ignore_source=True)
```

Or:

```bash
candump| cantools decode --j1939 --ignore-sa --dbc my.dbc
```

---

## ✅ Benefits

* Makes `cantools` useful for analyzing `candump` logs from tractors, sprayers, and other ISOBUS/J1939 devices.
* Removes the need to patch DBC files with every possible SA/DA combination.
* Aligns with J1939 decoding practices (as seen in tools like Vector CANalyzer or Wireshark).

---

## 🛠 Workaround (Current)

The current workaround is to:

* Manually duplicate message entries in the DBC for each SA.
* Write custom CAN ID parsing logic to match PGNs.

But this is tedious and brittle.

oak-tree, 2025-05-13:
Here are some possible implementation options:

1. Monkey patching using lookup tables.

1. Add a decorator to parse J1939 messages and automatically generate the necessary lookup tables.

1. Use a standard lookup table, assuming that for PGNs supporting multiple source/destination addresses, the DBC file represents them using 0xFE.

Code: 
```

###########################################
# monkey_patch_j1939.py
# Option 1: Monkey patching cantools
###########################################

import cantools
from cantools.j1939 import frame_id_unpack

def pgn_from_frame_id(frame_id):
    unpacked = frame_id_unpack(frame_id)
    if unpacked.pdu_format < 240:
        pdu_specific = 0
    else:
        pdu_specific = unpacked.pdu_specific

    return (unpacked.data_page << 16) | (unpacked.pdu_format << 8) | pdu_specific

# Patch function with alternative matching strategy
_original_get_message = cantools.database.can.Database.get_message_by_frame_id

def get_message_by_frame_id_patched(self, frame_id):
    # 1. Full CAN ID match
    msg = self._frame_id_to_message.get(frame_id & self._frame_id_mask)
    if msg:
        return msg

    unpacked = frame_id_unpack(frame_id)

    # 2. Wildcard source address (SA = 0xFE) only if PDU2
    if unpacked.pdu_format >= 240:
        frame_id_sa_wild = (frame_id & 0xFFFFFF00) | 0xFE
        msg = self._frame_id_to_message.get(frame_id_sa_wild & self._frame_id_mask)
        if msg:
            return msg

    # 3. Wildcard both SA and DA (for PDU1)
    if unpacked.pdu_format < 240:
        frame_id_sada_wild = (frame_id & 0xFFFF0000) | (0xFE << 8) | 0xFE
        msg = self._frame_id_to_message.get(frame_id_sada_wild & self._frame_id_mask)
        if msg:
            return msg

    return None

# Apply monkey patch
cantools.database.can.Database.get_message_by_frame_id = get_message_by_frame_id_patched

```

```

###########################################
# j1939_cantools_extension.py
# Option 2: Proper extension for PR
###########################################

from typing import Optional, Dict, Tuple
from cantools.j1939 import frame_id_unpack

class J1939ExtendedDatabase(cantools.database.can.Database):
    def __init__(self, *args, **kwargs):
        super().__init__(*args, **kwargs)
        self._build_j1939_lookup_tables()

    def _build_j1939_lookup_tables(self):
        self._pgn_to_message: Dict[int, cantools.database.can.message.Message] = {}
        self._pgn_sa_to_message: Dict[Tuple[int, int], cantools.database.can.message.Message] = {}
        self._pgn_sada_to_message: Dict[Tuple[int, int, int], cantools.database.can.message.Message] = {}

        for message in self._frame_id_to_message.values():
            if getattr(message, 'protocol', '').lower() != 'j1939':
                continue

            unpacked = frame_id_unpack(message.frame_id)
            pgn = pgn_from_frame_id(message.frame_id)
            sa = unpacked.source_address
            pf = unpacked.pdu_format
            ps = unpacked.pdu_specific

            self._pgn_to_message[pgn] = message
            self._pgn_sa_to_message[(pgn, sa)] = message

            if pf < 240:
                self._pgn_sada_to_message[(pgn, sa, ps)] = message

    def get_message_by_frame_id(self, frame_id: int, *, use_pgn_fallback: bool = False) -> Optional[cantools.database.can.message.Message]:
        try:
            return self._frame_id_to_message[frame_id & self._frame_id_mask]
        except KeyError:
            if not use_pgn_fallback or not hasattr(self, '_pgn_to_message'):
                raise

            unpacked = frame_id_unpack(frame_id)
            pgn = pgn_from_frame_id(frame_id)
            sa = unpacked.source_address
            pf = unpacked.pdu_format
            ps = unpacked.pdu_specific

            if pf < 240:
                msg = self._pgn_sada_to_message.get((pgn, sa, ps))
                if msg:
                    return msg

            msg = self._pgn_sa_to_message.get((pgn, sa))
            if msg:
                return msg

            return self._pgn_to_message.get(pgn)

```

```

###########################################
# fuzzy_wildcard_patch.py
# Option 3: Fuzzy wildcard matching without lookup tables
###########################################

import cantools
from cantools.j1939 import frame_id_unpack

def get_message_by_frame_id_fuzzy(self, frame_id):
    mask = self._frame_id_mask

    # 1. Exact match
    msg = self._frame_id_to_message.get(frame_id & mask)
    if msg:
        return msg

    unpacked = frame_id_unpack(frame_id)

    # 2. Wildcard source address (SA = 0xFE) only for PDU2
    if unpacked.pdu_format >= 240:
        frame_id_sa_fe = (frame_id & 0xFFFFFF00) | 0xFE
        msg = self._frame_id_to_message.get(frame_id_sa_fe & mask)
        if msg:
            return msg

    # 3. Wildcard both SA and DA (for PDU1)
    if unpacked.pdu_format < 240:
        frame_id_sada_fe = (frame_id & 0xFFFF0000) | (0xFE << 8) | 0xFE
        msg = self._frame_id_to_message.get(frame_id_sada_fe & mask)
        if msg:
            return msg

    return None

# Optional: apply instead of the basic patch
# cantools.database.can.Database.get_message_by_frame_id = get_message_by_frame_id_fuzzy

```

zariiii9003, 2025-05-13:
I know nothing about J1939, but maybe the `frame_id_mask` parameter of the Database class helps. Try setting `frame_id_mask=0x3ffff00`.