cantools/cantools #749: Decoding dbc file fails from 40.2.2 and upwards

nono313, 2025-08-22:
I have a dbc file that I am using to generate and parse CAN messages using cantools Python module.

Here is the dbc file:
```
VERSION "DBC file v1.0.0"

NS_ :
	NS_DESC_
	CM_
	BA_DEF_
	BA_
	VAL_
	CAT_DEF_
	CAT_
	FILTER
	BA_DEF_DEF_
	EV_DATA_
	ENVVAR_DATA_
	SGTYPE_
	SGTYPE_VAL_
	BA_DEF_SGTYPE_
	BA_SGTYPE_
	SIG_TYPE_REF_
	VAL_TABLE_
	SIG_GROUP_
	SIG_VALTYPE_
	SIGTYPE_VALTYPE_
	BO_TX_BU_
	BA_DEF_REL_
	BA_REL_
	BA_DEF_DEF_REL_
	BU_SG_REL_
	BU_EV_REL_
	BU_BO_REL_
	SG_MUL_VAL_

BS_:

BU_:

BO_ 2147483660 SelectPowerOut: 1 Vector__XXX
 SG_ SelectSecondOutput : 0|1@1+ (1,0) [0|0] "" Vector__XXX

BO_ 2147483661 RespSelectPowerOut: 1 Vector__XXX
 SG_ Status : 0|8@1+ (1,0) [0|0] "" Vector__XXX

CM_ BO_ 2147483660 "Select power output";
CM_ SG_ 2147483660 SelectSecondOutput "Bit used to select second power output (when 1)";
CM_ BO_ 2147483661 "Response to select power output";
BA_DEF_ SG_  "SPN" INT 0 524287;
BA_DEF_ BO_  "VFrameFormat" ENUM  "StandardCAN","ExtendedCAN","reserved","J1939PG";
BA_DEF_  "DatabaseVersion" STRING ;
BA_DEF_  "BusType" STRING ;
BA_DEF_  "ProtocolType" STRING ;
BA_DEF_  "DatabaseCompiler" STRING ;
BA_DEF_ BO_  "GenMsgCycleTime" INT 0 65535;
BA_DEF_DEF_  "SPN" 0;
BA_DEF_DEF_  "VFrameFormat" "J1939PG";
BA_DEF_DEF_  "DatabaseVersion" "DEMO PLUS";
BA_DEF_DEF_  "BusType" "";
BA_DEF_DEF_  "ProtocolType" "";
BA_DEF_DEF_  "DatabaseCompiler" "";
BA_DEF_DEF_  "GenMsgCycleTime" 0;
BA_ "ProtocolType" "J1939";
BA_ "BusType" "CAN";
BA_ "DatabaseCompiler" "CSS ELECTRONICS (WWW.CSSELECTRONICS.COM)";
BA_ "DatabaseVersion" "1.0.0";

```

I am decoding the messages with this line:
```
db = cantools.database.load_file(dbc_file_path)
message = db.decode_message(can_msg.arbitration_id, can_msg.data)
```

Everything works with cantools v40.2.1 but as soon as I update to v40.2.2, I get this error : 
```
    def decode_message(self,
                       frame_id_or_name: Union[int, str],
                       data: bytes,
                       decode_choices: bool = True,
                       scaling: bool = True,
                       decode_containers: bool = False,
                       allow_truncated:  bool = False,
                       force_extended_id: bool = False,
                       ) \
        -> DecodeResultType:
    
        """Decode given signal data `data` as a message of given frame id or
        name `frame_id_or_name`. Returns a dictionary of signal
        name-value entries.
    
        If `decode_choices` is ``False`` scaled values are not
        converted to choice strings (if available).
    
        If `scaling` is ``False`` no scaling of signals is performed.
    
        >>> db.decode_message(158, b'\\x01\\x45\\x23\\x00\\x11')
        {'Bar': 1, 'Fum': 5.0}
        >>> db.decode_message('Foo', b'\\x01\\x45\\x23\\x00\\x11')
        {'Bar': 1, 'Fum': 5.0}
    
        If `decode_containers` is ``True``, container frames are
        decoded. The reason why this needs to be explicitly enabled is
        that decoding container frames returns a list of ``(Message,
        SignalsDict)`` tuples which will cause code that does not
        expect this to misbehave. Trying to decode a container message
        with `decode_containers` set to ``False`` will raise a
        `DecodeError`.
        """
    
        if isinstance(frame_id_or_name, int):
            if force_extended_id or frame_id_or_name > 0x7FF:
                frame_id_or_name |= 0x80000000
>           message = self._frame_id_to_message[frame_id_or_name]
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
E           KeyError: 13

.venv/lib/python3.12/site-packages/cantools/database/can/database.py:583: KeyError
```

As I am using CSS editor, #748 may be linked to my issue.

zariiii9003, 2025-08-22:
try `message = db.decode_message(can_msg.arbitration_id, can_msg.data, force_extended_id=can_msg.is_extended_id)`

nono313, 2025-08-22:
Thanks. It's working. I guess it was added since 40.2.2.

closed 2025-08-22 by nono313