cantools/cantools #752: How to consider supporting Rust

lcwxz1989, 2025-09-22:
rust is a popular language than before, how to consider supporting Rust

juleq, 2025-09-22:
If you want to try adding rust support (I assume for code generation), you can have a look at c_source.py. You would fork cantools, copy c_source.py, name it rust_source.py and implement your rust generator there in a dimilar fashion. 

For integration, you also need to add a rust subparser ultimately.

When that is done you can write tests to make sure your generator code is covered and remains in good working order.

Lastly, you issue a pull request to this project.

It is a neat project to step into this.

trnila, 2025-09-22:
For DBC files, there’s already an existing project - [dbc-codegen](https://github.com/technocreatives/dbc-codegen) - written entirely in Rust. The code generation can be nicely integrated into `build.rs`. Just not sure what features it misses compared to `cantools`.